ftrias/TeensyDebug #4: Not building for Teensy 3.6

blackketter, 2020-09-15:
Alas, TeensyDebug doesn't seem to support Teensy 3.6 out of the box, building with this error below.  It looks like RAM_START and RAM_END are not defined for __MK66FX1M0__
I'm not familiar with the differences in the CPUs, is defining the appropriate RAM locations enough to get this working?
Thanks!
```
/Users/dean/src/TeensyDebug/src/gdbstub.cpp: In function 'int isValidAddress(uint32_t, int)':
/Users/dean/src/TeensyDebug/src/gdbstub.cpp:534:25: error: 'RAM_START' was not declared in this scope
   if (addr >= (uint32_t)RAM_START && addr <= (uint32_t)RAM_END) {
                         ^
/Users/dean/src/TeensyDebug/src/gdbstub.cpp:534:56: error: 'RAM_END' was not declared in this scope
   if (addr >= (uint32_t)RAM_START && addr <= (uint32_t)RAM_END) {
                                                        ^
/Users/dean/src/TeensyDebug/src/gdbstub.cpp:539:30: error: 'FLASH_START' was not declared in this scope
   else if (addr >= (uint32_t)FLASH_START && addr <= (uint32_t)FLASH_END) {
                              ^
/Users/dean/src/TeensyDebug/src/gdbstub.cpp:539:63: error: 'FLASH_END' was not declared in this scope
   else if (addr >= (uint32_t)FLASH_START && addr <= (uint32_t)FLASH_END) {
```

ftrias, 2020-09-15:
This was never tested on a 3.5 or 3.6. I don't know what hardware support these CPUs have for breakpoint. If you are not going to use breakpoints, setting RAM_START/END FLASH_START/END should be enough. If you want to use breakpoints, then more research is required.

blackketter, 2020-09-15:
Great, I’ll give it a shot.  I’m just looking to break in when it seems stuck in an infinite loop, for now...

blackketter, 2020-09-15:
One question: I see how RAM_START and FLASH_START match the values in the .ld files for the cpus in question, but I'm not sure how to calculate the RAM_END and FLASH_END.  

FLASH_END seems to be straightforward, though the end value is the address immediately after the last flash byte.

RAM_END seems to be a much larger value than the actual RAM size and is set to the end of a block (ie. ends with FFFF).  Is this the total addressable RAM space (independent of the actual amount of RAM)?

Excuse my ignorance, I'm trying to wrap my head around this stuff...  Thanks in advance!

ftrias, 2020-09-15:
RAM_END should be the last real memory address. The purpose of these
constants is to avoid doing a memory access to non-existent memory, which
will crash the cpu.

firelizzard18, 2021-02-22:
> I don't know what hardware support these CPUs have for breakpoint.

The Teensy 3.x family all use Kinetis K series microcontrollers, so I imagine they have similar breakpoint support. Either way, I'd be happy to investigate support for the 3.6 at least (because that's what I have).

I added RAM and FLASH boundaries for the Teensy 3.6:

```c
#ifdef __MK66FX1M0__
#define FLASH_START ((void*)0x0)
#define FLASH_END ((void*)0x000FFFFF)
#define RAM_START ((void*)0x1FFF0000)
#define RAM_END   ((void*)0x2002FFFF)
#endif
```

These are based on the linker script for the 3.6, `mk66fx1m0.ld`:

```
MEMORY
{
	FLASH (rx) : ORIGIN = 0x00000000, LENGTH = 1024K
	RAM  (rwx) : ORIGIN = 0x1FFF0000, LENGTH = 256K
}
```

I modified `boards.local.txt` to add the 3.6. I can successfully launch the `simple_breakpoint` example with Teensyduino and GDB > "Take over serial". However, I'm getting some strange results. The breakpoint appears to be triggering in `HardwareSerial::addToSerialEventsList`. The example in the README (explicitly call `halt()`) halts correctly. But in either case, attempting to step hangs something.

firelizzard18, 2021-02-22:
I have successfully debugged the following sketch:

```C++
#include "TeensyDebug.h"
#pragma GCC optimize ("O0")

int mark = 0;

DEBUGRUN
void test_function() {
  mark++;
}

void setup() {
  debug.begin();
}

DEBUGRUN
void loop() {
  test_function();
  Serial.println(mark);
  delay(1000);
}
```

GDB still initially breaks during `HardwareSerial::addToSerialEventsList` for some reason, but I can add a breakpoint with `b test_function` and continue and it works as expected.

firelizzard18, 2021-02-22:
`debug.setBreakpoint(test_function)` works, once. When I upload and debug, GDB stops at `test_function`, but when I continue it does not stop again. I can however Ctrl+C and add a breakpoint manually, which works.

However, if I `debug.setCallback(break_me)` (as in the `simple_breakpoint` example), GDB no longer works. Since the callback is normally set to `process_onbreak`, I can see why overwriting it would cause issues. Is `setCallback` supposed to be able to work alongside GDB?

ftrias, 2021-02-22:
Thanks for the suggestions for Teensy 3.6. I'll see about adding them.

I don't know why it breaks at addToSerialEventsList().

As you probably suspected, debug.setCallback(...)  overrides the default
breakpoint handling and GDB will no longer work. It is for use when you
have some other external debugger.

firelizzard18, 2021-02-22:
Everything I've tested works, except that `debug.setBreakpoint(<function>)` only hits once. LMK if there's something specific you'd like me to test.

ftrias, 2021-05-25:
Closing as fixed. Please reopen if that is not so.

closed 2021-05-25 by ftrias