ftrias/TeensyDebug #7: [Errno 2] No such file or directory: 'README.md'

lukehutch, 2021-05-17:
I get `[Errno 2] No such file or directory: 'README.md'` when I try to install TeensyDebug on Linux:

```
[luke@localhost ~]$ cd /tmp
[luke@localhost tmp]$ git clone https://github.com/ftrias/TeensyDebug.git
Cloning into 'TeensyDebug'...
remote: Enumerating objects: 389, done.
remote: Counting objects: 100% (108/108), done.
remote: Compressing objects: 100% (77/77), done.
remote: Total 389 (delta 65), reused 67 (delta 29), pack-reused 281
Receiving objects: 100% (389/389), 47.82 MiB | 9.49 MiB/s, done.
Resolving deltas: 100% (242/242), done.
[luke@localhost tmp]$ cd TeensyDebug/extras/
[luke@localhost extras]$ ls
boards.local.txt  install-mac.command  platform.local.txt
build.bat         install-windows.txt  teensy_debug
install-linux.sh  Makefile             teensy_debug.exe
[luke@localhost extras]$ bash install-linux.sh 
Install GDB in Teensyduino
Teensyduino found in /opt/arduino-1.8.13/
Teensyduino tools in /opt/arduino-1.8.13/hardware/tools/
Teensyduino custom library in /home/luke/Arduino/libraries/TeensyDebug/
Is this OK?
[y/N]? y
Create /opt/arduino-1.8.13/hardware/teensy/avr/boards.local.txt
Create /opt/arduino-1.8.13/hardware/teensy/avr//platform.local.txt
Copy teensy_debug to /opt/arduino-1.8.13/hardware/tools/
Copy README.md to /home/luke/Arduino/libraries/TeensyDebug/
[Errno 2] No such file or directory: 'README.md'
Press Enter to close

[luke@localhost extras]$ cd ..
[luke@localhost TeensyDebug]$ bash extras/install-linux.sh 
Install GDB in Teensyduino
Teensyduino found in /opt/arduino-1.8.13/
Teensyduino tools in /opt/arduino-1.8.13/hardware/tools/
Teensyduino custom library in /home/luke/Arduino/libraries/TeensyDebug/
Is this OK?
[y/N]? y
Already installed. Updating.
File /opt/arduino-1.8.13/hardware/teensy/avr/boards.local.txt already exists. Would you like to overwrite it? 
[y/N]? y
Create /opt/arduino-1.8.13/hardware/teensy/avr/boards.local.txt
Create /opt/arduino-1.8.13/hardware/teensy/avr//platform.local.txt
Copy teensy_debug to /opt/arduino-1.8.13/hardware/tools/
Copy README.md to /home/luke/Arduino/libraries/TeensyDebug/
[Errno 2] No such file or directory: 'README.md'
Press Enter to close
```

lukehutch, 2021-05-17:
The installer only works if I move everything from `extras` and `src` to the root directory of the project before running it.

ftrias, 2021-05-19:
Good catch. This is a by-product of restructuring the directories to march the newer library structure, but forgetting to change teensy_debug. I just updated the source code for teensy_debug to fix this problem.

ftrias, 2021-05-25:
Updated git with fix and closing.

closed 2021-05-25 by ftrias