ftrias/TeensyDebug #8: Including TeensyDebug.h makes incompatible changes to `Serial`

lukehutch, 2021-05-17:
After including `TeensyDebug.h`, the following canonical code

```
  while(!Serial);
  Serial.begin(19200);
```

gives these errors:

```
Test_Teensy: In function 'void setup_main()':
Test_Teensy:75: error: no match for 'operator!' (operand type is 'Debug')
   while(!Serial);
         ^
/home/luke/Arduino/Test_Teensy/Test_Teensy.ino:75:9: note: candidate: operator!(bool) <built-in>
/home/luke/Arduino/Test_Teensy/Test_Teensy.ino:75:9: note:   no known conversion for argument 1 from 'Debug' to 'bool'
/home/luke/Arduino/Test_Teensy/Test_Teensy.ino:76:21: warning: invalid conversion from 'int' to 'Stream*' [-fpermissive]
   Serial.begin(19200);
                     ^
In file included from /home/luke/Arduino/Test_Teensy/Test_Teensy.ino:9:0:
/home/luke/Arduino/libraries/TeensyDebug/TeensyDebug.h:134:7: note:   initializing argument 1 of 'int Debug::begin(Stream*)'
   int begin(Stream *device = NULL);
       ^
no match for 'operator!' (operand type is 'Debug')
```

lukehutch, 2021-05-17:
(Obviously this is in "Take over serial" mode)

ftrias, 2021-05-18:
This is an  interesting oversight. I think the solution is to  add the following to the class Debug (TeensyDebug.h)

operator bool() { return true; }

lukehutch, 2021-05-18:
I tried adding this line to `class Debug`, and it didn't fix the problem.

ftrias, 2021-05-19:
Try adding this as well to class Debug.

  int begin(int baud) { return 1; }

lukehutch, 2021-05-19:
That second line (`begin(int baud)`) works (and it works without the `bool()` definition), at least for the two `Serial` lines that I provided.

ftrias, 2021-05-25:
Updated git with fix and closing bug

closed 2021-05-25 by ftrias