ftrias/TeensyDebug #9: Hardware debugging?

lukehutch, 2021-05-19:
I am finding it is very hard to debug anything nontrivial with TeensyDebug -- the breakpoints and stepping are just not working for my code (I am compiling with `-g -O0`). Teensy seems to just hang after the first debugging step, most of the time I try to debug my code using TeensyDebug.

I assume this is because of the software workarounds that are needed (replacing code in-place), due to the hardware "debug enable bit" not being set by the firmware. I saw a thread about that somewhere on the forums. I realize it is nontrivial to get software debugging working well.

What is the status on getting the firmware modified to enable hardware debugging support?

ftrias, 2021-05-19:
Paul (main Teensy developer) has expressed resistance to enabling hardware on numerous occasions, which requires a hardware change, so I don't think that's going to happen. Are you using Teensy 3 or 4? Stepping is unreliable over branching and function returns and is best done at the assembly level. But breakpoints should work.

lukehutch, 2021-05-19:
I'm on Teensy 4.1.

I saw some strange things, such as after stopping in the debugger on `halt()`, when I ran `i lo` (`info locals`, if I remember right, to see local variables), the constructor of one of the local variables that had not been initialized yet was actually run (maybe this is a "feature" of gdb, I don't know, but this constructor had major side effects). Stepping was also unreliable even when there was no branching.

I guess I'll try to work only with breakpoints, not with stepping, and I'll see if I can get results that way.

I'll close this for now, and I'll report any specific issues I find. Thanks for the info on hardware debugging.

closed 2021-05-19 by lukehutch