ftrias/TeensyDebug #10: Compiling TeensDebug with -Os not possible

danie1kr, 2021-12-26:
Hi,
I wanted to minimize the footprint of all code including TeensyDebug with the -Os option.
This results in the following error messages:
```
TeensyDebug.cpp: 1279:6: error: prototype for 'void Debug::setCallback(void (*)())' does not match any in class 'Debug
 void Debug*: setCallback(void (*c)()) { callback = c; }
 
Error compiling libraries
Debug build failed for project 'winston-teensy'
TeensyDebug.cpp:38: In file included from
 
TeensyDebug.h: 139:8: error: candidate is: void Debug::setCallback(void (*)())
   void setCallback(void (*c)())
```

The compiler command lines were:
```
Using library TeensyDebug version 0.0.1 in folder "file:///C:/Users/daniel/Documents/Arduino/libraries/TeensyDebug"
"C:\Program Files (x86)\Arduino\hardware\teensy/../tools/arm/bin/arm-none-eabi-g++" -c -Og -g -Wall -ffunction-sections -fdata-sections -nostdlib -MMD -std=gnu++14 -fno-exceptions -fpermissive -fno-threadsafe-statics -felide-constructors -Wno-error=narrowing -mthumb -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-d16 -D__IMXRT1062__ -DTEENSYDUINO=155 -DARDUINO=108016 -DARDUINO_TEENSY41 -DF_CPU=600000000 -DUSB_DUAL_SERIAL -DLAYOUT_US_ENGLISH "-IC:\Users\daniel\AppData\Local\Temp\VMBuilds\winston-teensy\teensy41\Debug/pch" -I"C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy4" -I"c:\program files\microsoft visual studio\2022\community\common7\ide\extensions\itelwv1j.wbw\Micro Platforms\default\debuggers\VM_DBG" -I"C:\Users\daniel\Documents\Arduino\libraries\WebSockets2_Generic\src" -I"C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\NativeEthernet\src" -I"C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\FNET\src" -I"C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\SPI" -I"C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\SD\src" -I"C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\SdFat\src"  -Os  -I"D:\proj\dev\uc\winston\winston\winston-teensy" -I"D:\proj\dev\uc\winston\winston\winston-teensy\..\libwinston" -I"D:\proj\dev\uc\winston\winston\winston-teensy\..\winston"  -I"D:\proj\dev\uc\winston\winston\winston-teensy" -I"D:\proj\dev\uc\winston\winston\winston-teensy\..\libwinston" -I"D:\proj\dev\uc\winston\winston\winston-teensy\..\winston" -DVM_DEBUG_BREAKPAUSE -DVM_DBT=VM_DBT_GENERIC_OBJECT -DVM_DEBUGGER_SOFT_TRANSPORT_WRITER=Serial -DVM_DEBUGGER_SOFT_TRANSPORT=Serial -DVM_DEBUG_BANDWIDTH_THROTTLE_MS=100 -DVM_DEBUG -DVM_DEBUG_ENABLE=1 -DVM_DBT_GENERIC_OBJECT=21 -DVM_DBT_NO_SERIAL=20 -DVM_DBT_MS430_SERIAL_=19 -DVM_DBT_SERIALUSB=18 -DVM_DBT_USBAPI=17 -DVM_DBT_NET_UDP=16 -DVM_DBT_HARDWARESERIAL3=15 -DVM_DBT_HARDWARESERIAL2=14 -DVM_DBT_HARDWARESERIAL1=13 -DVM_DBT_CDCSerialClass=12 -DVM_DBT_COSA=11 -DVM_DBT_Uart=10 -DVM_DBT_NET_CONSOLE=9 -DVM_DBT_TTYUART=8 -DVM_DBT_USBSERIAL=7 -DVM_DBT_USART=6 -DVM_DBT_UART=5 -DVM_DBT_TEENSY=4 -DVM_DBT_USB=3 -DVM_DBT_FASTSERIAL=2 -DVM_DBT_SOFTWARESERIAL=1 -DVM_DBT_HARDWARESERIAL=0 -DTL_NAMESPACE=std -DWINSTON_PLATFORM_TEENSY -DTCB_SPAN_NAMESPACE_NAME=std -I"C:\Users\daniel\Documents\Arduino\libraries\TeensyDebug\src" "C:\Users\daniel\Documents\Arduino\libraries\TeensyDebug\src\gdbstub.cpp" -o "C:\Users\daniel\AppData\Local\Temp\VMBuilds\winston-teensy\teensy41\Debug\TeensyDebug\gdbstub.cpp.o"
 
gdbstub.cpp: In function void gdb_init(Stream*)
gdbstub.cpp: 1186:36: warning: invalid conversion from 'void (*)()' to 'void (*)()' [-fpermissive]
   debug.setCallback(process_onbreak)
 
gdbstub.cpp:19: In file included from
TeensyDebug.h:139: note    initializing argument 1 of void Debug  setCallback(void (*)())
   void setCallback(void (*c)())
"C:\Program Files (x86)\Arduino\hardware\teensy/../tools/arm/bin/arm-none-eabi-g++" -c -Og -g -Wall -ffunction-sections -fdata-sections -nostdlib -MMD -std=gnu++14 -fno-exceptions -fpermissive -fno-threadsafe-statics -felide-constructors -Wno-error=narrowing -mthumb -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-d16 -D__IMXRT1062__ -DTEENSYDUINO=155 -DARDUINO=108016 -DARDUINO_TEENSY41 -DF_CPU=600000000 -DUSB_DUAL_SERIAL -DLAYOUT_US_ENGLISH "-IC:\Users\daniel\AppData\Local\Temp\VMBuilds\winston-teensy\teensy41\Debug/pch" -I"C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy4" -I"c:\program files\microsoft visual studio\2022\community\common7\ide\extensions\itelwv1j.wbw\Micro Platforms\default\debuggers\VM_DBG" -I"C:\Users\daniel\Documents\Arduino\libraries\WebSockets2_Generic\src" -I"C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\NativeEthernet\src" -I"C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\FNET\src" -I"C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\SPI" -I"C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\SD\src" -I"C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\SdFat\src"  -Os  -I"D:\proj\dev\uc\winston\winston\winston-teensy" -I"D:\proj\dev\uc\winston\winston\winston-teensy\..\libwinston" -I"D:\proj\dev\uc\winston\winston\winston-teensy\..\winston"  -I"D:\proj\dev\uc\winston\winston\winston-teensy" -I"D:\proj\dev\uc\winston\winston\winston-teensy\..\libwinston" -I"D:\proj\dev\uc\winston\winston\winston-teensy\..\winston" -DVM_DEBUG_BREAKPAUSE -DVM_DBT=VM_DBT_GENERIC_OBJECT -DVM_DEBUGGER_SOFT_TRANSPORT_WRITER=Serial -DVM_DEBUGGER_SOFT_TRANSPORT=Serial -DVM_DEBUG_BANDWIDTH_THROTTLE_MS=100 -DVM_DEBUG -DVM_DEBUG_ENABLE=1 -DVM_DBT_GENERIC_OBJECT=21 -DVM_DBT_NO_SERIAL=20 -DVM_DBT_MS430_SERIAL_=19 -DVM_DBT_SERIALUSB=18 -DVM_DBT_USBAPI=17 -DVM_DBT_NET_UDP=16 -DVM_DBT_HARDWARESERIAL3=15 -DVM_DBT_HARDWARESERIAL2=14 -DVM_DBT_HARDWARESERIAL1=13 -DVM_DBT_CDCSerialClass=12 -DVM_DBT_COSA=11 -DVM_DBT_Uart=10 -DVM_DBT_NET_CONSOLE=9 -DVM_DBT_TTYUART=8 -DVM_DBT_USBSERIAL=7 -DVM_DBT_USART=6 -DVM_DBT_UART=5 -DVM_DBT_TEENSY=4 -DVM_DBT_USB=3 -DVM_DBT_FASTSERIAL=2 -DVM_DBT_SOFTWARESERIAL=1 -DVM_DBT_HARDWARESERIAL=0 -DTL_NAMESPACE=std -DWINSTON_PLATFORM_TEENSY -DTCB_SPAN_NAMESPACE_NAME=std -I"C:\Users\daniel\Documents\Arduino\libraries\TeensyDebug\src" "C:\Users\daniel\Documents\Arduino\libraries\TeensyDebug\src\TeensyDebug.cpp" -o "C:\Users\daniel\AppData\Local\Temp\VMBuilds\winston-teensy\teensy41\Debug\TeensyDebug\TeensyDebug.cpp.o"
```

Adding 
```
#pragma GCC push_options
#pragma GCC optimize("O0")
#pragma GCC pop_options
```
 to both files resolved the issue for me. 

Maybe, you can look into this.

Best Regards
Daniel

ftrias, 2021-12-28:
Those are strange errors. What version of Arduino and Teensyduino are you using? Did you download TeensyDebug from here or are you using the one that comes with Teensyduino? How exactly are you compiling this?

danie1kr, 2021-12-28:
I also find it strange.
However:

- Arduino IDE is 1.8.16 but I use Visual Studio 2022 with Visual Micro (which internally uses Arduino IDE as you might know).
- TeensyDuino is the most recent version compatible with 1.8.16
- TeensyDebug is from git directly, master-branch.zip Downloaded 9.12.2021

I noticed the same warning (`invalid conversion from void (*)()' to 'void (*)()`) also for other Arduino libraries, e.g. SPI or SD. But it did not give any compile errors. I have not yet checked the proper functioning of the code as the project is not in the proper state for now.

Best Regards
Daniel

ftrias, 2021-12-29:
I just committed a change that may fix this. It was caused by using different optimization levels between the function definitions and saving function pointers. If you have time, please update and try this.

danie1kr, 2022-01-04:
I took the latest master.zip and still have the same result:
```
Using library TeensyDebug version 0.0.1 in folder "file:///C:/Users/daniel/Documents/Arduino/libraries/TeensyDebug"
"C:\Program Files (x86)\Arduino\hardware\teensy/../tools/arm/bin/arm-none-eabi-g++" -c -Og -g -Wall -ffunction-sections -fdata-sections -nostdlib -MMD -std=gnu++14 -fno-exceptions -fpermissive -fno-threadsafe-statics -felide-constructors -Wno-error=narrowing -mthumb -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-d16 -D__IMXRT1062__ -DTEENSYDUINO=155 -DARDUINO=108016 -DARDUINO_TEENSY41 -DF_CPU=600000000 -DUSB_DUAL_SERIAL -DLAYOUT_US_ENGLISH "-IC:\Users\daniel\AppData\Local\Temp\VMBuilds\winston-teensy\teensy41\Debug/pch" -I"C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy4" -I"c:\program files\microsoft visual studio\2022\community\common7\ide\extensions\itelwv1j.wbw\Micro Platforms\default\debuggers\VM_DBG" -I"C:\Users\daniel\Documents\Arduino\libraries\WebSockets2_Generic\src" -I"C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\NativeEthernet\src" -I"C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\FNET\src" -I"C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\SPI" -I"C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\SD\src" -I"C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\SdFat\src"  -Os  -I"D:\proj\dev\uc\winston\winston\winston-teensy" -I"D:\proj\dev\uc\winston\winston\winston-teensy\..\libwinston" -I"D:\proj\dev\uc\winston\winston\winston-teensy\..\winston"  -I"D:\proj\dev\uc\winston\winston\winston-teensy" -I"D:\proj\dev\uc\winston\winston\winston-teensy\..\libwinston" -I"D:\proj\dev\uc\winston\winston\winston-teensy\..\winston" -DVM_DEBUG_BREAKPAUSE -DVM_DBT=VM_DBT_GENERIC_OBJECT -DVM_DEBUGGER_SOFT_TRANSPORT_WRITER=Serial -DVM_DEBUGGER_SOFT_TRANSPORT=Serial -DVM_DEBUG_BANDWIDTH_THROTTLE_MS=100 -DVM_DEBUG -DVM_DEBUG_ENABLE=1 -DVM_DBT_GENERIC_OBJECT=21 -DVM_DBT_NO_SERIAL=20 -DVM_DBT_MS430_SERIAL_=19 -DVM_DBT_SERIALUSB=18 -DVM_DBT_USBAPI=17 -DVM_DBT_NET_UDP=16 -DVM_DBT_HARDWARESERIAL3=15 -DVM_DBT_HARDWARESERIAL2=14 -DVM_DBT_HARDWARESERIAL1=13 -DVM_DBT_CDCSerialClass=12 -DVM_DBT_COSA=11 -DVM_DBT_Uart=10 -DVM_DBT_NET_CONSOLE=9 -DVM_DBT_TTYUART=8 -DVM_DBT_USBSERIAL=7 -DVM_DBT_USART=6 -DVM_DBT_UART=5 -DVM_DBT_TEENSY=4 -DVM_DBT_USB=3 -DVM_DBT_FASTSERIAL=2 -DVM_DBT_SOFTWARESERIAL=1 -DVM_DBT_HARDWARESERIAL=0 -DTL_NAMESPACE=std -DWINSTON_PLATFORM_TEENSY -DTCB_SPAN_NAMESPACE_NAME=std -I"C:\Users\daniel\Documents\Arduino\libraries\TeensyDebug\src" "C:\Users\daniel\Documents\Arduino\libraries\TeensyDebug\src\gdbstub.cpp" -o "C:\Users\daniel\AppData\Local\Temp\VMBuilds\winston-teensy\teensy41\Debug\TeensyDebug\gdbstub.cpp.o"
 
gdbstub.cpp: In function void gdb_init(Stream*)
gdbstub.cpp: 1186:36: warning: invalid conversion from 'void (*)()' to 'void (*)()' [-fpermissive]
   debug.setCallback(process_onbreak)
 
gdbstub.cpp:19: In file included from
TeensyDebug.h:139: note    initializing argument 1 of void Debug  setCallback(void (*)())
   void setCallback(void (*c)())
"C:\Program Files (x86)\Arduino\hardware\teensy/../tools/arm/bin/arm-none-eabi-g++" -c -Og -g -Wall -ffunction-sections -fdata-sections -nostdlib -MMD -std=gnu++14 -fno-exceptions -fpermissive -fno-threadsafe-statics -felide-constructors -Wno-error=narrowing -mthumb -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-d16 -D__IMXRT1062__ -DTEENSYDUINO=155 -DARDUINO=108016 -DARDUINO_TEENSY41 -DF_CPU=600000000 -DUSB_DUAL_SERIAL -DLAYOUT_US_ENGLISH "-IC:\Users\daniel\AppData\Local\Temp\VMBuilds\winston-teensy\teensy41\Debug/pch" -I"C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy4" -I"c:\program files\microsoft visual studio\2022\community\common7\ide\extensions\itelwv1j.wbw\Micro Platforms\default\debuggers\VM_DBG" -I"C:\Users\daniel\Documents\Arduino\libraries\WebSockets2_Generic\src" -I"C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\NativeEthernet\src" -I"C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\FNET\src" -I"C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\SPI" -I"C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\SD\src" -I"C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\SdFat\src"  -Os  -I"D:\proj\dev\uc\winston\winston\winston-teensy" -I"D:\proj\dev\uc\winston\winston\winston-teensy\..\libwinston" -I"D:\proj\dev\uc\winston\winston\winston-teensy\..\winston"  -I"D:\proj\dev\uc\winston\winston\winston-teensy" -I"D:\proj\dev\uc\winston\winston\winston-teensy\..\libwinston" -I"D:\proj\dev\uc\winston\winston\winston-teensy\..\winston" -DVM_DEBUG_BREAKPAUSE -DVM_DBT=VM_DBT_GENERIC_OBJECT -DVM_DEBUGGER_SOFT_TRANSPORT_WRITER=Serial -DVM_DEBUGGER_SOFT_TRANSPORT=Serial -DVM_DEBUG_BANDWIDTH_THROTTLE_MS=100 -DVM_DEBUG -DVM_DEBUG_ENABLE=1 -DVM_DBT_GENERIC_OBJECT=21 -DVM_DBT_NO_SERIAL=20 -DVM_DBT_MS430_SERIAL_=19 -DVM_DBT_SERIALUSB=18 -DVM_DBT_USBAPI=17 -DVM_DBT_NET_UDP=16 -DVM_DBT_HARDWARESERIAL3=15 -DVM_DBT_HARDWARESERIAL2=14 -DVM_DBT_HARDWARESERIAL1=13 -DVM_DBT_CDCSerialClass=12 -DVM_DBT_COSA=11 -DVM_DBT_Uart=10 -DVM_DBT_NET_CONSOLE=9 -DVM_DBT_TTYUART=8 -DVM_DBT_USBSERIAL=7 -DVM_DBT_USART=6 -DVM_DBT_UART=5 -DVM_DBT_TEENSY=4 -DVM_DBT_USB=3 -DVM_DBT_FASTSERIAL=2 -DVM_DBT_SOFTWARESERIAL=1 -DVM_DBT_HARDWARESERIAL=0 -DTL_NAMESPACE=std -DWINSTON_PLATFORM_TEENSY -DTCB_SPAN_NAMESPACE_NAME=std -I"C:\Users\daniel\Documents\Arduino\libraries\TeensyDebug\src" "C:\Users\daniel\Documents\Arduino\libraries\TeensyDebug\src\TeensyDebug.cpp" -o [... cut ...]
TeensyDebug.cpp: 1273:6: error: prototype for 'void Debug::setCallback(void (*)())' does not match any in class 'Debug
 void Debug*: setCallback(void (*c)()) { callback = c; }
 
TeensyDebug.cpp:38: In file included from
 
TeensyDebug.h: 139:8: error: candidate is: void Debug::setCallback(void (*)())
   void setCallback(void (*c)())
```
I noticed there are still some optimization pragmas in the code:
```c
#pragma GCC push_options
#pragma GCC optimize ("O0")
```
Maybe there is a way to skip them all or to guard both files with the pragma. What do you think?

ftrias, 2022-01-04:
You have to "pull" from github to get the changes. master.zip hasn't been
updated.

danie1kr, 2022-01-05:
Good morning, I downloaded the code and it had your latest commits in place.
However, with the repository cloned directly, the issues persist (see above). Only way for me to get around them, is to guard the whole file as described.