ftrias/TeensyDebug #12: Running debug.begin stop Teensy from running properly [VSCODE][Platformio]

jparrish88, 2022-01-10:
Hi @ftrias , 

I really love the idea of what you created here. I am doing my best to get it operational, however after multiple nights of trying different things. I'm a bit lost. 

I am running:
Windows 10
VS Code 1.63.2
PlatformIO (core 5.2.4)
Teensyduino/framwork-arduinoteensy (1.154)
Teensy 4.1 
GCCARMNOEEABI 1.504 (5.4.1)

I've ran both the launch configuration edits on the readme as well as and separately this script I found from the below posting (changed from teensy40 to teensy41):
https://community.platformio.org/t/full-support-for-teensy-4-0-build-test-debug-with-platformio/9249/11
```
[env:teensy41]
platform = teensy
board = teensy41
framework = arduino
build_type = debug
lib_deps = 
    https://github.com/ftrias/TeensyDebug/archive/refs/heads/master.zip
; activate Dual USB just as README says
build_flags =
    -D USB_DUAL_SERIAL
upload_protocol = teensy-cli
debug_port = COM7
debug_tool = custom
debug_load_mode = manual
debug_server = 
debug_init_cmds =
  target extended-remote $DEBUG_PORT
  $INIT_BREAK
  define pio_reset_run_target
  interrupt
  tbreak loop
  continue
  end
  define pio_restart_target
  echo Restart is undefined for now.
  end

debug_init_break =
```
I've also stripped out the complete portion below the build flags. 

On build I get a gdbstub.cpp.o and a teensydebug.cpp.o along with a libTeensyDebug.a file. 

BUT to my dismay, no matter how I build the program I typically have an issue where the hex file, once uploaded, ends up causing the teensy to never open up either serial port (in fact it looks dead or bricked). Luckily the button always brings it back alive. 

That said, I posted it as an issue here. Simply because if I remove the debug.begin(SerialUSB1) and reprogram I get the dual serials to appear --- BUT I can't seem to connect with them over VSCODE or straight from the GDB tool. 

Additionally, when I remove the debug.begin line I get weirdness where the teensy consistently disconnects. ( as if every few seconds it was performing a reset). 

This is my complete code (as found from your Readme:
```
#include <Arduino.h>
#include "TeensyDebug.h"
#pragma GCC optimize ("O0")

int mark = 0;

void test_function() {
  mark++;
}

void setup() {
  // Use the first serial port as you usually would
  Serial.begin(19200);

[... 113 lines cut ...]
Compiling .pio\build\teensy41\FrameworkArduino\usb_rawhid.c.o
Compiling .pio\build\teensy41\FrameworkArduino\usb_seremu.c.o
Compiling .pio\build\teensy41\FrameworkArduino\usb_serial.c.o
Compiling .pio\build\teensy41\FrameworkArduino\usb_serial2.c.o
Compiling .pio\build\teensy41\FrameworkArduino\usb_serial3.c.o
Compiling .pio\build\teensy41\FrameworkArduino\usb_touch.c.o
Compiling .pio\build\teensy41\FrameworkArduino\yield.cpp.o
Archiving .pio\build\teensy41\libFrameworkArduino.a
Linking .pio\build\teensy41\firmware.elf
Calculating size .pio\build\teensy41\firmware.elf
Checking size .pio\build\teensy41\firmware.elf
Building .pio\build\teensy41\firmware.hex
teensy_size: Memory Usage on Teensy 4.1:
teensy_size:   FLASH: code:38308, data:5544, headers:8368   free for files:8074244
teensy_size:    RAM1: variables:17216, code:35976, padding:29560   free for local variables:441536
teensy_size:    RAM2: variables:24736  free for malloc/new:499552
================================================= [SUCCESS] Took 4.29 seconds =================================================
Terminal will be reused by tasks, press any key to close it.
```

I didn't install TeensyDebug though, as I am unsure where the correct directory is to install it for VSCODE and platformio. I found a directory in some of the Platformio folders, but there is also no Tools directory. I would install it, if needed, but the python script you made doesn't seem to show anything different with the build process happening; it more looks like where and how gdb is automatically opened and sent the programming commands. What I'm not sure is if there needs to be some special tool that uploads the program to begin with && how to even make that work with VSCODE. (It seems like that's not the case). 

Essentially I'm hoping you can help me figure out what's going on? Why would teensy not even run properly when "debug.begin" is left in? Why would teensy just act like it's restarting over and over again when debug.begin is removed? 

When debug.begin is removed and I try to connect to it with the gdb program, I get an error saying that the remote connection was closed. Windows also makes it sound like there was some sort of hardware reset occurring.

ftrias, 2022-01-10:
I am not too familiar with PlatformIO but it seems a lot of people use it so I did a little digging.

One issue is that the default config doesn't compile for dual USB. See: https://github.com/platformio/platform-teensy/issues/65

I tried it with this platform.ini and it works for me:

```
[env:teensy40]
platform = teensy
board = teensy40
framework = arduino

build_type = debug
; https://github.com/platformio/platform-teensy/issues/65
build_unflags = -DUSB_SERIAL
build_flags = -DUSB_DUAL_SERIAL
debug_port = /dev/cu.usbmodem61684903 
debug_tool = custom
debug_load_mode = manual
debug_server = 
debug_init_cmds =
  target extended-remote $DEBUG_PORT
  $INIT_BREAK
  define pio_reset_run_target
  interrupt
  tbreak loop
  continue
  end
  define pio_restart_target
  echo Restart is undefined for now.
  end

debug_init_break =
```

Can you try this and let me know if it works?

jparrish88, 2022-01-12:
Unfortunately, it doesn't seem to work. By that, I mean that it "Fails" in the same way, where I no longer have any COM ports opening and it essentially acts as if it is bricked. 

I do have new information. I've built with both the Arduino IDE and the Visual Teensy (which works with VSCODE). Both of those solutions work and allow for debug as intended. 

So I believe that the problem, lie squarely on PlatformIO's build. I'm unsure if that means that your files would need to change to meet their library needs OR (hopefully) that there's just some setting in PlatformIO that causes a problem. 

Additionally, I am able to get Dual USB serial running with the original ini file I showed, when disabling TeensyDebug features. Meaning that I don't need to use the unflags line. 

From here, do you mind if I ask what you would suggest? Should I post on PlatformIO's community instead. I mean it's seems like your lib works (wonderfully, I might add), just not with PlatformIO so far [At least, for me]. I'd really like to get PlatformIO features running, as most of my projects are built off that.

ftrias, 2022-01-12:
I ran it on PlatformIO and compiled and debugged a program. Is there something off about your install? Do you need to upgrade something? Asking on the PlatformIO forum is probably a good idea. Please report back if you find out something helpful.