ftrias/TeensyDebug #13: PlatformIO instructions are incomplete

fieryWaters, 2022-02-12:
Hello, 
I'm trying to get your library running, but the instructions are hard to follow. I modified the platformio.ini file to closely match what was given in the example, but there are no other instructions. 
`You need to change the debug_port item to match whatever port your systems assigns.`
This step is ambiguous to me. I would really appreciate some step by step instructions on what I need to do to get it working in platformIO.

I'm using Visual Studio code on MacOS. Could that be the issue? 

I appologize for not giving a more concrete issue, but without step by step instructions there is no concrete point for me to say "I went wrong here." I humbly recommend the devs actually install everything from nothing and document each step they take. A lot is lost in translation with all the intimate knowledge y'all have of the library you wrote. 

I would really love to make use of this library, please help!

ftrias, 2022-02-12:
When you connect your Teensy compiled with dual serials to the computer, it gets allocated two serial ports. But it's impossible to know which ones ahead of time. So you just look in /dev/tty* to see what files suddenly appear. Those are the serial ports. One of them is for regular output and one for the debugging. Again, impossible to tell without some experimentation. So maybe you can add a few print statements to your  Teensy program and see where they show up. You can use "miniterm" or similar command-line programs to monitor the data sent to a device.

It should work fine on a Mac. That's my dev platform.

k4kfh, 2022-03-16:
So I'm on Linux and I'm also having issues with setup. I watched /dev while I plugged in my Teensy, and it appears to be using `/dev/ttyACM0` and `/dev/ttyACM1`, but I tried both and can't seem to get it to work. I'm using VS Code and PlatformIO. Any suggestions? If someone can help me get this going, I'll do my best to write some more detailed documentation, because I am so excited that this project exists!

DaniGarciaLopez, 2022-03-25:
I'm having the same problem on linux. I can correctly see `/dev/ttyACM0` and `/dev/ttyACM1` and compilation is sucessfull but the breakpoint is not triggered. I tried several months ago and I gave up, looks like `build_unflags = -DUSB_SERIAL` makes no difference. Any help is welcome!

DaniGarciaLopez, 2022-03-25:
Hi @ftrias. I'm using Ubuntu 20.04 with PlatformIO 2.4.3. If I use Serial.print(...) I receive it in ACM0, so I guess I should set `debug_port = /dev/ttyACM1`, right?

I was using the project I created several months ago to try `build_unflags = -DUSB_SERIAL`, but now I've just created a new project from scratch and I when I build it it throws errors with TeensyDebug regarding 'sprintf' and 'strlen'. What is the preferred way to compile TeensyDebug in PlatformIO?

Previously, as @[maxgerhardt](https://community.platformio.org/u/maxgerhardt) [mentioned here](https://community.platformio.org/t/full-support-for-teensy-4-0-build-test-debug-with-platformio/9249/10), I was using `lib_deps = https://github.com/ftrias/TeensyDebug/archive/refs/heads/master.zip`. I also tried adding the library to `/lib` with no luck.

Thanks!

rogerjames99, 2022-11-22:
On Linux platforms the /dev/ttyACM* ports are allocated in the order they are detected by the usb cdc kernel driver. If you have a directory on your system called /dev/serial/by-id you can also find symlinks to the ports in there. The names of these symlinks have the subsidiary device id of the ports encoded in them.

lrwxrwxrwx 1 root root 13 Nov 22 08:43 usb-Teensyduino_Dual_Serial_12205920-if00 -> ../../ttyACM1
lrwxrwxrwx 1 root root 13 Nov 22 08:43 usb-Teensyduino_Dual_Serial_12205920-if02 -> ../../ttyACM0

The one ending in 00 is the Serial device. The one ending in 02 is the SerialUSB1 device.

The assigment to ACM0 or ACM1 can be pretty arbritary, it depends on which one is detected first.