ftrias/TeensyDebug #25: Does not work with freeRTOS

yamaan93, 2023-11-15:
I was very relieved when I saw this project existed because I found it ridiculous that there is no hardware debugger built into the teensy considering how powerful and capable it is, however, once you have freeRTOS running, the debugger seizes to operate. It will work up until the scheduler starts, however after that, breakpoints are ignored.

ftrias, 2023-11-15:
I'm guessing that FreeRTOS probably takes over the timer interrupts. You'll have to figure out how to use a FreeRTOS timer instead. I'm not familiar with that.

yamaan93, 2023-11-18:
Do you have any suggestions for what exactly needs to be replaced with a software interrupt? For example I tried following the instructions in this comment and adding a +1:
[image]
but nothing changed.